arduino/Arduino #10700: [Library Manager] Could you please add arduino-slack-library to the library manager

witnessmenow, 2020-08-26:
Hello,

Could you please add arduino-slack-library to the library manager:
https://github.com/witnessmenow/arduino-slack-api

Thanks,
Brian

cmaglie, 2020-08-31:
Hi @witnessmenow sorry for the delay.

You named your library `ArduinoSlackAPI` the prefix `Arduino*` is reserved for the libraries made and maintained by Arduino, that's why we do not accept new libraries with this prefix.

Can you change the name of your library? `SlackAPI` or anything else will work.

To do this remember to:
1. change the name in `library.properties`
2. commit & push
3. delete the latest git tag and redo it on master

Thanks!

witnessmenow, 2020-08-31:
Hello @cmaglie 

I've updated the library.properties, bumped the version and tagged a new release. I also deleted the old tag.

Thanks!

cmaglie, 2020-08-31:
Done, thanks!

closed 2020-08-31 by cmaglie