arduino/Arduino #10702: [Library Manager] Please add KIM Library

ncassarkineis, 2020-08-27:
Please add [KIM library](https://github.com/Kineis/ArduinoKim) to Library Manager.
Thanks and Regards.

per1234, 2020-08-28:
Hi @ncassarkineis. 
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

William-at-kineis, 2020-08-28:
Hi @per1234,

We have just released   https://github.com/Kineis/ArduinoKim/releases/tag/ARDUINO_KIM1_HW2_V3.0.0 .
Is this syntax fine for you or do you need a `v3.0.0` release as well?

cmaglie, 2020-08-31:
Done.

closed 2020-08-31 by cmaglie