arduino/Arduino #10704: [Library Manager] Please add M5Core2 Library

Gitshaoxiang, 2020-08-28:
Please add M5Core library to Library Manager.

[M5Core2](https://github.com/m5stack/M5Core2)

Thanks and Regards.

per1234, 2020-08-28:
Hi @Gitshaoxiang. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Gitshaoxiang, 2020-08-28:
I forgot to add it, sorry. I have now added the release.

Gitshaoxiang, 2020-08-31:
hi.   it not add to the Arduino Library Manager yet right?  we did some update about the library.   now we released the tag again.   could you mind check it again?

cmaglie, 2020-08-31:
It's ok, now the library is published.

closed 2020-08-31 by cmaglie

Gitshaoxiang, 2020-09-01:
our library json file has some mistake description....now we fixed.    how we could update the new info to the published lib? do i need to release a new tag ?  could you help us to solve this problem.    thanks.

per1234, 2020-09-01:
@Gitshaoxiang Arduino Library Manager does not use library.json. That is the metadata file for Platformio. Arduino Library Manager only uses library.properties.

FYI, [Platformio also supports library.properties](https://docs.platformio.org/en/latest/librarymanager/creating.html), so you  may be able to remove library.json and use a single metadata file (library.properties) for both in order to avoid the burden of maintaining two parallel metadata files.

Gitshaoxiang, 2020-09-01:
yep, thank you reply... but the mistake description is from the `library.properties`  file.    ....how we could update it to the Arduino Library Manager?  (wrong board name in the information.... ,  but now we fixed)

Gitshaoxiang, 2020-09-02:
hello? any response ?

cmaglie, 2020-09-02:
Hi @Gitshaoxiang 

in theory, you should produce another release of the library (say `0.0.2`) since each release has his own metadata and the IDE will display metadata of the latest release.

BTW, for this time, I've edited the fields in the index, so it should be ok now.

Gitshaoxiang, 2020-09-03:
thank you very much!