arduino/Arduino #10705: Request for adding EasyMorse Library

milador, 2020-08-28:
Can you please add the EasyMorse Library?

It's available at : [https://github.com/milador/EasyMorse](https://github.com/milador/EasyMorse)

Thanks

per1234, 2020-08-28:
Hi @milador. 
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

milador, 2020-08-28:
Hi @per1234,

I created the release and it should be good to go. 

[https://github.com/milador/EasyMorse/releases](https://github.com/milador/EasyMorse/releases)

Thanks

cmaglie, 2020-08-31:
Done.

closed 2020-08-31 by cmaglie