arduino/Arduino #10708: avr-gcc-4.8.1-arduino5-i686-mingw32.zip

haoran33798780, 2020-08-29:
The board I'm using is digispark, so I've added http://digistump.com/package_digistump_index.json to the preference ,then i select digistump avr boards from the board manager,and it begin to download,but after a while there was an error warning:
download error: http://downloads.arduino.cc/tools/avr-gcc-4.8.1-arduino5-i686-mingw32.zip 
java.lang.RuntimeException: java.lang.Exception: http://downloads.arduino.cc/tools/avr-gcc-4.8.1arduino5-i686-mingw32.zip 
       at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:181)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception:  http://downloads.arduino.cc/tools/avr-gcc-4.8.1-arduino5-i686-mingw32.zip 
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:113)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:174)
	... 1 more
Caused by: java.lang.Exception: Incomplete download
	at cc.arduino.utils.network.FileDownloader.readStreamCopyTo(FileDownloader.java:283)
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:247)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
	... 5 more
please help

per1234, 2020-08-29:
Hi @haoran33798780. This issue be caused by a temporary network outage. Please try installing digistump avr boards from Boards Manager again.

After that, please comment here to let us know what the result was.

haoran33798780, 2020-08-29:
> Hi @haoran33798780. This issue be caused by a temporary network outage. Please try installing digistump avr boards from Boards Manager again.
> 
> After that, please comment here to let us know what the result was.

thanks for 

> Hi @haoran33798780. This issue be caused by a temporary network outage. Please try installing digistump avr boards from Boards Manager again.
> 
> After that, please comment here to let us know what the result was.

I have tried for many times,but it still doesn't work.

haoran33798780, 2020-08-29:
by the way i can download directly from http://downloads.arduino.cc/tools/avr-gcc-4.8.1-arduino5-i686-mingw32.zip ,but i cannot download it in board manager

per1234, 2020-09-28:
> i can download directly from ...

@haoran33798780 are you downloading to the same drive as the location shown in the Arduino IDE at the line following **File > Preferences > More preferences can be edited directly in the file**. I wonder if it could be caused by running out of disk space?

per1234, 2021-03-17:
Closing due to lack of response.

@haoran33798780  if you'll provide the requested information I'll be happy to re-open this issue.

closed 2021-03-17 by per1234