arduino/Arduino #10711: [Library Manager] Please add cmdArduino library

freaklabs, 2020-08-31:
Hi. I've created a library that I'd like to request to be added to the library manager. It can be found here:
https://github.com/freaklabs/cmdArduino

Thank you.
Akiba

per1234, 2020-08-31:
Hi @freaklabs. Every library library added to the Library Manager index must contain a `library.properties` file. Your library instead contains a `libraries.properties` file.

You should rename the file and create a new tag/release or redo the existing one.

Thanks!

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

freaklabs, 2020-08-31:
Hi. Sorry about that. I just renamed the file and redid the release. 
Thanks.
Akiba

cmaglie, 2020-08-31:
Done.

closed 2020-08-31 by cmaglie