arduino/Arduino #10731: Why does Serial.print ("\n\n\nAborted !!!\n\n\n"); that the Sketch can not always be downloaded?

bgma, 2020-09-06:
Hi,
Why does `Serial.print ("\n\n\nAborted !!! \n\n\n");` that the Sketch can not always be uploaded? But it works sometimes!
I have seen this several times. Now it has happened again.
When it does not work, it is enough to remove 1 of the 3 !!!.
It is always possible to compile but not to upload.
After a long time, the error message "An error occurred while uploading the sketch"

/With kind regards
 Bengt-Görana

[sketch_sep06a.txt](https://github.com/arduino/Arduino/files/5179855/sketch_sep06a.txt)

[image]

per1234, 2020-09-06:
Closing as duplicate of https://github.com/arduino/Arduino/issues/392

Please see that issue for an explanation of the cause. You can check Google to find a lot of information about how to burn a modern bootloader to your Mega to make this not happen.

closed 2020-09-06 by per1234

bgma, 2020-09-06:
Tankyou!