arduino/Arduino #10744: [Library manager] My library EMailSender not update automatically

xreef, 2020-09-09:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
Hi all,
my library inserted on issue #10505 not automatically updated, Did I make any declaration errors?
https://github.com/xreef/EMailSender

Thanks Bye

per1234, 2020-09-10:
The problem is that you forgot to update the "version" value in library.properties before making your v2.0.1 release.

You can see that it's the same in both the v2.0.0:
https://github.com/xreef/EMailSender/blob/v2.0.0/library.properties#L2
and v2.0.1 tags:
https://github.com/xreef/EMailSender/blob/v2.0.1/library.properties#L2

The units of the Library Manager index are the Git tags, but everthing else is based on the library.properties metadata file. The actual name of the tag is completely ignored. This is a bit confusing because tags are often named according to version numbers (as is very sensible), but they can have any arbitrary name.

So the indexer job compared the library.properties version value of each of your repository's tags to the existing entries in the index for the library and found that there were no new versions available. So it sees your v2.0.1 release, but ignores it.

You will need to update the library.properties version value and then either make a new release or redo the existing release.

closed 2020-09-10 by per1234

xreef, 2020-09-10:
Sossy and thanks per1234, your help is precious.