arduino/Arduino #10746: How to add  library.properties to library

yannisant, 2020-09-09:
Hello,

I am trying to upload  the library for the [DFRobot_BME280] to the online editor and I get an error message about the library.properties file missing. Can you please help? Do I generate this file? Or can I find it?

Kind regards,

Yannis

matthijskooijman, 2020-09-09:
Here's the reference documentation for libraries: https://arduino.github.io/arduino-cli/latest/library-specification/

There might be better, tutorial-style documentation elsewhere, but I"m not sure where so hopefully this will at least get you started.

per1234, 2020-09-25:
Since Matthijs was kind enough to help you out and we already have an issue for the excellent suggestion of adding information about library.properties to a tutorial at https://github.com/arduino/Arduino/issues/4413, I'm going to go ahead and close this.

closed 2020-09-25 by per1234