arduino/Arduino #10749: Murata SCA100T -library for inclinometer readings

CsCrazy85, 2020-09-11:
https://github.com/CsCrazy85/SCA100T

First release of the SCA100T -library, which allows a user to easily get angle, temperature and acceleration values from the Murata SCA100T-inclinometer sensors. Library offers a basic functionality for trivial tasks such as extracting a value from a register and converting it into a usable value. Also there's a demonstration on how to use Arduino SPISettings object when user wants to change SPI bus speed on the fly, and sensors self-test features are shown.

Examples folder contains 5 basic usage code examples to start working with this library.

Library is tested with SCA100TD02 sensor, but it should work fine with SCA100TD01 sensor as well. Only difference is the sensitivity of the sensor, which affects the calculation formula, but it's handled within the library.

https://github.com/CsCrazy85/SCA100T/releases/tag/v1.0.0

cmaglie, 2020-09-14:
Hi @CsCrazy85 
I've added the library to the indexer but there is a problem: you named the `library.properties` file as `library.properties.txt`.

To fix this:

1. rename `library.properties.txt` to `library.properties`
2. commit & push
3. delete the git tag `v1.0.0` and re-do it on master

after doing the above, the library will be automatically published within an hour, so I'm closing this issue as solved for now. Feel free to comment here again if you need further assistance.

closed 2020-09-14 by cmaglie

CsCrazy85, 2020-09-14:
> Lähetetty: maanantai 14. syyskuuta 2020 14:17
> Vastaanottaja: arduino/Arduino<mailto:Arduino@noreply.github.com>
> Kopio: Mika Väyrynen<mailto:mika_va@hotmail.com>; Mention<mailto:mention@noreply.github.com>

Hi!

Oh yeah I forgot the extension. It’s now fixed and re-released with version v1.0.0 tag. Thanks for your help!

-Mika