arduino/Arduino #10750: Website Forum - something went wrong with 2-factor Auth and now cannot log in

Gerriko, 2020-09-11:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

The title should be self explanatory.

Website forum page had asked me to set up 2-factor. I say yes and about to choose Google's Authenticator option. Then realise that I don't have my phone with me. Close down web page.

Now I am in "limbo land" as Sign In button just refreshes screen.

I can log/sign in and log/sign out of Arduino.cc. This just affects the forum page.

BallscrewBob, 2020-09-21:
This was also mentioned in the topic https://forum.arduino.cc/?topic=698486#msg4695043

BallscrewBob, 2020-09-21:
@Gerriko Have you tried the usual steps of cleaning the browser cache / cookies etc.
It would seem that whilst setting up 2FA there should be no changes until some form of final button / selection option is made or that closing a browser should eliminate any prior selections if the final item in not processed.

Gerriko, 2020-09-22:
Yes I tried the usual steps of clearing browser cache and removing cookies etc. But this appeared not to work. I say "appeared" because it may well have worked as observed below, but I never went beyond the main forum screen when I could not see that I had been signed in.

So, I applied the suggestion of turning off 2FA on Google's side (as per suggestion in the link supplied above). That at least removed the request for 2FA. Which is good.

I then went through the normal login routine while on the main forum page. It still did not show that I had been logged in. I refreshed the page and it still did not update (still had the sign in icon). I then clicked on a forum post. Now when this page loaded it showed that I was logged in. I returned to the main forum page and this also showed that I was logged in.

Hope that helps. I am using Mozilla Firefox on Linux OS.

BallscrewBob, 2020-09-22:
There is a current issue open where users may be logged in to some parts of the web site but show as NOT logged in when moving to another part and having to sign in again.

I found that going to the BLOG section and coming back to forum would allow me to sign in properly.
Arduino are aware of the issue of multiple logins being required and appear to be working on the problem.
But glad you are at least back into the forum,