arduino/Arduino #10759: [Library Manager]

aptinexuser, 2020-09-14:
Please add APTINEX_DAC to the library manager. 
https://github.com/aptinex/Aptinex_DAC

per1234, 2020-09-14:
Hi @aptinexuser. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

aptinexuser, 2020-09-14:
Hi
I have created a realese for this library

aptinexuser, 2020-09-16:
Hi, @per1234 
How long does it take to publish this library to the library manager?

per1234, 2020-09-18:
Hello. Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this. There may be a delay of some hours after that for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient. Thanks!

cmaglie, 2020-09-23:
Hi @aptinexuser, sorry for the long delay.

I've added the library to the indexer but there is a problem, your `library.properties` has `depends=*` that is not a valid value

```
JOB 3350 - 2020/09/23 22:03:25 Scraping https://github.com/aptinex/Aptinex_DAC.git
JOB 3350 - 2020/09/23 22:03:27 Checking out tag: 1.0.0
JOB 3350 - 2020/09/23 22:03:27 Error syncing library: Error generating library from repo: Invalid 'depends' field: *
```

if your library does not depend on other libraries you can fix it:

1. remove the `depends=*` line from your library.properties
2. commit & push
3. delete and redo the `1.0.0` git tag

once done, the library will be automatically fetched and published by our indexer, within an hour.
I'm going ahead and close the issue, BTW feel free to comment here if you need further assistance.

closed 2020-09-23 by cmaglie