arduino/Arduino #10761: When board folder is in Arduino/hardware, it compiles but it does not upload

ekawahyu, 2020-09-16:
I use STM32 platform with Arduino IDE 1.8.13 on Ubuntu 18.04, and installing the 3rd party hardware support from here:

https://github.com/stm32duino/Arduino_Core_STM32

I tested this Arduino IDE on Windows, Linux, and macOS and all give me the same result as follows. I can confirm that there is no issue with STM32 hardware support package nor OS related. I did many experiments on this.

1. With the default Board Manager installation, Arduino can compile and upload just fine.
2. When I move the variant folder from .arduino15 to Arduino/hardware, Arduino IDE can compile, but fail to upload.
3. When I move the tools folder from .arduino15 to Arduino/hardwware, Arduino IDE can compile, but fail to upload.
4. When I move both variant and tools folders from .arduino15 to Arduino/hardware, Arduino IDE can compile, but fail to upload.

For point (2), (3), and (4) above, they all failed at EXECUTING upload command with {runtime.tools.<vendor_tool>.path} under the uploader rules/definition. I managed to debug a little bit from the error messages and I figured that {runtime.tools.<vendor_tool>.path} is all correctly translated for when I put the tools folder under .arduino15 or Arduino/hardware. This is a bit strange for me, because, if the compiler can see where the tools folder is, by translating {runtime.tools.<vendor_tool>.path} correctly, then why does the uploader failed to see where the tools folder is?

I can live with point (1) for now for code development, but the code can get accidentally removed through uninstaller. So, really, what I need is to have at least the variant folder to remain under Arduino/hardware.

Is this a known bug with Arduino IDE? I am using the latest version, 1.8.13, but I am open to downgrade it, if there is any version that works with variant folder under Arduino/hardware.

There is a workaround if I have to have it under Arduino/hardware. The {runtime.tools.<vendor_tool>.path} can be temporarily replaced with {runtime.hardware.path} plus the absolute path down to the upload script. But again, this is troublesome because everytime I need to push/merge my changes, I need to remind myself to put it back to {runtime.tools}

Is it a bug in Arduino IDE, or anything I do wrong with the setup?

jandrassy, 2020-09-17:
my hardware folder in the sketchbook folder is https://github.com/jandrassy/my_boards. 
I don't use a custom variants folder, but I added one as a copy of STM32:PILL_F103XX and I changed in boards.txt `BluePill128kswd.build.variant=custombp`, the stm32CubeProg.bat starts the upload. I will try later on Linux.

ekawahyu, 2020-09-17:
> my hardware folder in the sketchbook folder is https://github.com/jandrassy/my_boards.
> I don't use a custom variants folder for stm32, but to test I added one as a copy of STM32:PILL_F103XX and I changed in boards.txt `BluePill128kswd.build.variant=custombp`. The stm32CubeProg.bat starts the upload. I will try later on Linux.

@jandrassy Would you mind to create a branch of what you did so that I can test it on my end please?

jandrassy, 2020-09-17:
> @jandrassy Would you mind to create a branch of what you did so that I can test it on my end please?

https://github.com/jandrassy/my_boards/tree/stm32_variants_test

ekawahyu, 2020-09-21:
@jandrassy I have managed to add custom variants within the hardware folder, and also custom DFU tool. Thank you for providing the branch for me to look at. That is very helpful.

I am still struggling with one more thing: having custom core within Arduino/hardware. Have you done this before? All I need is to modify the USB CDC stack.

https://github.com/SnapBloks/SnapBloks

jandrassy, 2020-09-22:
> @jandrassy I have managed to add custom variants within the hardware folder, and also custom DFU tool. Thank you for providing the branch for me to look at. That is very helpful.
>
> I am still [...]

if you only need a different initialization of USB, then I think you can do it in variants.
if you want to change the code in core, then you will have to copy all the core source files. you can still reference the tools.

ekawahyu, 2020-09-22:
I wanted to do it in variant, but the problem is... when I enable the USB serial, the core takes the source code from .arduino15. How to let Arduino IDE knows that a particular file (i.e. core/arduino/stm32/usb/cdc/usbd_cdc_if.c) or the entire USB CDC stack must be built from Arduino/hardware instead?

jandrassy, 2020-09-22:
> I wanted to do it in variant, but the problem is... when I enable the USB serial, the core takes the source code from .arduino15. How to let Arduino IDE knows that a particular file (i.e. [...]

if you want to change a file in core, you must copy the whole `cores/arduino` folder.
what is what you want to change in `usbd_cdc_if.c`?

ekawahyu, 2020-09-22:
Just one liner in `usbd_cdc_if.c` to intercept USB serial communication.

This is the branch develop with additional cores/arduino and it does not build. It says missing recipe... meaning that I need to copy and modified platform.txt as well?

https://github.com/SnapBloks/SnapBloks/tree/develop

sank-tech, 2020-09-23:
View it on GitHub

jandrassy, 2020-09-24:
move cores/arduino into stm32

ekawahyu, 2020-09-24:
did try it, but it didn't work either. just in case you wanted to see the complete error messages:

```
Using board 'SnapBloksLED' from platform in folder: /Users/macos/Documents/Arduino/hardware/SnapBloks/stm32
Using core 'arduino' from platform in folder: /Users/macos/Documents/Arduino/hardware/SnapBloks/stm32
Detecting libraries used...
recipe.preproc.macros pattern is missing

Error compiling for board SnapBloks LED.
```
I wish there was a way to debug this by having more verbose error messages.

closed 2020-09-25 by per1234

jandrassy, 2020-09-25:
yes, copy platform.txt too