arduino/Arduino #10764: Arduino 1.8.13 issue with upload and Mega2560

PickyBiker, 2020-09-17:
Please see this link for details 
[](https://forum.arduino.cc/index.php?topic=705481.0)
Environment: Win 10,  Arduino 1.8.13,  Genuine Arduino Mega2560
The upload function seems to start the mega2560 briefly then restarts it. 
In the example sketch it results in a double output of "Hello" instead of a single output. 
This problem does not exist in 1.8.12 and earlier versions. 
`// example mega2560 problem
void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println("Hello");
  }

void loop()
{
}`

per1234, 2020-09-30:
Some discussion on this subject on the forum:
https://forum.arduino.cc/?topic=703890

PickyBiker, 2020-09-30:
Thanks Per.