arduino/Arduino #10766: When encountering ''Incorrect IDE installation folder", the output is frustratingly silent on what needs to be fixed

brewmanz, 2020-09-18:
The two reasons (`Your copy of the IDE is installed in a subfolder of your settings folder.\nPlease move the IDE to another folder.` & `Your copy of the IDE is installed in a subfolder of your sketchbook.\nPlease move the IDE to another folder.`) do not actually state what specifically needs to be fixed. 

Maybe this is okay for seasoned Arduino developers, but why not actually tell the newbie developer what the conflicting settings are?

One solution could be to change `BaseNoGui.java` code like
```
    if (isIDEInstalledIntoSketchbookFolder()) {
      showError(tr("Incorrect IDE installation folder"), tr("Your copy of the IDE is installed in a subfolder of your sketchbook.\nPlease move the IDE to another folder."), 10);
    }
```
into
```
    if (isIDEInstalledIntoSketchbookFolder()) {
      if (PreferencesData.has("sketchbook.path")){
       showWarning(I18n.format("sketchbook.path=<{0}>", PreferencesData.get("sketchbook.path")), I18n.format("runtime.ide.path=<{0}>", PreferencesData.get("runtime.ide.path")), null);
      }
      showError(tr("Incorrect IDE installation folder"), tr("Your copy of the IDE is installed in a subfolder of your sketchbook.\nPlease move the IDE to another folder."), 10);
    }
```

per1234, 2020-09-25:
Thanks for your nice suggestion @brewmanz!

See https://github.com/arduino/Arduino/issues/10511 for an alternate suggestion for improving this experience for the user.

Other related issues:
- https://github.com/arduino/Arduino/issues/10572
- https://github.com/arduino/Arduino/issues/7280
- https://github.com/arduino/Arduino/issues/4995
- https://github.com/arduino/Arduino/issues/2719

matthijskooijman, 2020-11-20:
@per1234, you said "See #10511 for an alternate suggestion", but IIUC, that issue just suggests to *fix* the error message, since it is not shown. Here, @brewmanz suggests that the error message is shown, but is not verbose enough.

As for the suggestion done in the first post here, AFAIU that would cause two popups, one with just some paths, and then one explaining that there is a problem? That does not sound ideal, since at the first popup, the user has no idea what he's looking at? It would probably be better to have just a single popup like now, but *add* to that the problematic paths?

brewmanz, 2020-11-20:
@matthijskooijman I agree my solution is not ideal, but IMHO is a step in the right direction, and enabled me to work out WHY there was a problem, not merely than one existed without any clue of what needed fixing. I would be very happy for someone with more experience in this area to provide an improved solution.

gipsayan, 2020-11-20:
@matthijskooijman as you indicated under #10982, it is the same issue as #10511 and #10766. The warning text should sound something like "Please ensure that (1) there is a subfolder "sketchbook" under the Arduino installation folder and (2) that the sketchbook.path entry in the /.arduino15/preferences.txt file contains the full path of the sketchbook folder, for example sketchbook.path=/home/user/Arduino/sketchbook"
Users with some experience can then manually change folder names and locations as long as the matter, and the potential bug, is clear. But at least there will be a reference working version.

brewmanz, 2020-11-20:
Note that just stating what the path needs to look like is not sufficient. 
Without showing what the bad path actually was, I would have been scratching my head, as there were MULTIPLE configuration files and I was initially looking at, and fixing, THE WRONG ONE.