arduino/Arduino #10767: Library Manager: don't reset the "Type" to "All" after updating a library

SlickNet, 2020-09-18:
Updating individual libraries is a bit of a pain, but it is made worse when after selecting Type=Updatable, and clicking update on one library, the UI changes back to Type=All  after that one update.    Then you have to set it back to "Updateable"

Tested in 1.8.13

lhalf, 2021-04-08:
Hi - I'm a new contributor and thought I could try and implement a fix for this. I think I've found where this functionality is: Arduino/app/src/cc/arduino/contributions/libraries/ui/DropdownUpdatableLibrariesItem.java

Specifically:
`public class DropdownUpdatableLibrariesItem implements DropdownItem<ContributedLibraryReleases> {

  @Override
  public Predicate<ContributedLibraryReleases> getFilterPredicate() {
    return new Predicate<ContributedLibraryReleases>() {
      @Override
      public boolean test(ContributedLibraryReleases lib) {
        Optional<ContributedLibrary> mayInstalled = lib.getInstalled();
        if (!mayInstalled.isPresent()) {
          return false;
        }
        return !lib.getLatest().equals(mayInstalled.get());
      }
    };
  }

  @Override
  public String toString() {
    return tr("Updatable");
  }

}`

And under: Arduino/app/src/cc/arduino/contributions/libraries/ui/DropdownAllLibraries.java

`public class DropdownAllLibraries implements DropdownItem<ContributedLibraryReleases> {

  public String toString() {
    return tr("All");
  }

  @Override
  public Predicate<ContributedLibraryReleases> getFilterPredicate() {
    return x -> true;
  }

}`

Now these are the implementations, but after an update I assume a default will be selected again to "All". Any suggestions on where the default selection might be made?

cmaglie, 2021-04-26:
The fix for this issue is already on `master` and will be released with the IDE 1.8.14.

closed 2021-04-26 by cmaglie