arduino/Arduino #10768: [Feature Request] Library Manager: Allow an "update all" or selecting multiple libraries to update

SlickNet, 2020-09-18:
Updating libraries one by one can be tedious if you have a large set of libraries to update at any time.   Please provide a way to have the software iterate over all or a set of them to do the update, instead of having to select each one of them and deal with laggy UI.

per1234, 2020-09-18:
Thanks for your feature request @SlickNet. We already have a feature request for the "update all" component you requested: https://github.com/arduino/Arduino/issues/4438, so lets use the existing issue for that request and make this one solely for the "selecting multiple libraries to update" part of your request, which I don't see an existing request for.

It is possible to achieve both requests via a feature that provides a list of updatable libraries, which can be individually selected or deselected, then all selected libraries installed in a batch.

sank-tech, 2020-09-19:
View it on GitHub