arduino/Arduino #10772: [Library Manager] Please add Arduino-Scheduler library to the library manager

Robokishan, 2020-09-19:
hey I have developed a library and it is my first contribution to the opensource community.
Please add this arduino scheduler to the arduino library manager https://github.com/Robokishan/Arduino-Scheduler

per1234, 2020-09-19:
Hi @Robokishan. Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

After that's done, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Robokishan, 2020-09-19:
I have changed name and add tag

per1234, 2020-09-19:
Unfortunately, there's still a problem. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value used in your library has already been taken:
https://github.com/arduino-libraries/Scheduler

If you would still like your library added to the index, you'll need to change the `name` value to a unique name (check Library Manager to see if the name is already taken), then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag (you can delete releases and tags via the GitHub web interface if you don't want to do it from the command line).

After that is done, please comment here to let us know we can proceed. Thanks!

Robokishan, 2020-09-19:
Done name changed to Taskrunner 
https://github.com/Robokishan/Taskrunner

Robokishan, 2020-09-19:
So can you tell me approximately when it will get approve to get in the library manager

per1234, 2020-09-19:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this. There may be a delay of some hours after that for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

Robokishan, 2020-09-19:
Ok Thank you i am just curious regarding this since it is my first push in the Arduino community. Sorry for the inconvenience.

cmaglie, 2020-09-23:
Done, sorry for the long delay.

closed 2020-09-23 by cmaglie

Robokishan, 2021-06-25:
I have released new library version can you please update ?

https://github.com/Robokishan/Taskrunner/blob/master/library.properties