arduino/Arduino #10774: [Library Manager] Add MorsDuino library to then library manager

GGorAA, 2020-09-20:
Please add the library [MorsDuino](https://github.com/GGorAA/MorsDuino) to the Library Manager.

GGorAA, 2020-09-20:
I think this issue will be closed as soon as library appears in Arduino Library Manager?

per1234, 2020-09-20:
Yes:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

cmaglie, 2020-09-23:
Done.

@GGorAA you tagged the `1.1` and `2.0-beta1` but you forget to update the version in `library.properties`.

```
JOB 3362 - 2020/09/23 22:03:43 Checking out tag: 1.0
JOB 3362 - 2020/09/23 22:03:43 path=github.com/GGorAA, archivename=MorseDuino-1.0.0

JOB 3362 - 2020/09/23 22:03:44 Checking out tag: 1.1
JOB 3362 - 2020/09/23 22:03:47 Release MorseDuino:1.0.0 already loaded, skipping

JOB 3362 - 2020/09/23 22:03:47 Checking out tag: 2.0-beta1
JOB 3362 - 2020/09/23 22:03:47 Release MorseDuino:1.0.0 already loaded, skipping
```

Please update the version in `library.properties` (and push it on github) before tagging a new release otherwise the release will be ignored.

closed 2020-09-23 by cmaglie

GGorAA, 2020-09-24:
Thanks for adding it to Library Manager!

Yes, I will fix those issues about `library.properties`

GGorAA, 2020-09-24:
Issues fixed