arduino/Arduino #10779: [Library manager] ABB Aurora Inverter library (Arduino, esp32 and esp8266)

xreef, 2020-09-22:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
Source: https://github.com/xreef/ABB_Aurora_Solar_Inverter_Library
Tag: https://github.com/xreef/ABB_Aurora_Solar_Inverter_Library/releases/tag/v1.0.1

It's a complete library to interfacing via rs485 the ABB Aurora inverter series

Documentation
ENG: https://www.mischianti.org/2020/08/20/abb-aurora-pv-inverter-library-for-arduino-esp8266-and-esp32/
ITA: https://www.mischianti.org/it/2020/08/20/libreria-inverter-abb-aurora-pv-per-arduino-esp8266-and-esp32/

Bye Renzo

per1234, 2020-09-22:
Hi @xreef. Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

xreef, 2020-09-22:
Hi per1234,
I fixed It, thanks

Source: https://github.com/xreef/ABB_Aurora_Solar_Inverter_Library
Tag: https://github.com/xreef/ABB_Aurora_Solar_Inverter_Library/releases/tag/v1.0.2

Bye Renzo

cmaglie, 2020-09-23:
Done.

closed 2020-09-23 by cmaglie