arduino/Arduino #10793: Please add library PWMMotorControl

ArminJo, 2020-09-24:
Hello,

I prepared an new Arduino compatible library.
https://github.com/ArminJo/PWMMotorControl

Could you add it to your list of libraries?

Thank you.
Armin

per1234, 2020-09-24:
Hello @ArminJo. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

Thanks!

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

ArminJo, 2020-09-24:
Uuups.
Thanks , I forgot it.
Now the release is ready.

ArminJo, 2020-10-06:
Hi,
is there a reason for not adding this library into your library set???

per1234, 2020-10-06:
No reason at all. It will be added for sure.

cmaglie periodically comes through to clear all the outstanding Library Manager inclusion requests at once. You happened to submit this request only a few hours after the last batch addition happened, so you're unlucky enough to be experiencing the maximum delay.

So we'll get to it soon, please be patient. Thanks!

ArminJo, 2020-10-06:
Thanks, so I must hurry with my next one :-)

closed 2020-10-06 by cmaglie