arduino/Arduino #10795: [Documentation] ArduinoModbus Server, wrong return value for configure* functions

wayofka, 2020-09-25:
In documentation for ArduinoModbus library, all configure function for Server have wrong return value in documentation.
For example [https://www.arduino.cc/en/ArduinoModbus/ModbusServerconfigureDiscreteInputs](https://www.arduino.cc/en/ArduinoModbus/ModbusServerconfigureDiscreteInputs)
The function allocate memory for Moddus registers and coils.
```
Returns
0 on success, 1 on failure 
```
Return should be ( according to  source code )
-1  Invalid argument 
0 on failure 
1 on success

Same for all configure function

Prath06, 2020-11-02:
I want to contribute on this.

per1234, 2020-11-02:
@Prath06 I appreciate your interest in contributing to the Arduino project. Unfortunately, the documentation content is not hosted in a public repository, so it will not be possible for you to assist with this.

The same goes for the other issue you commented on (https://github.com/arduino/Arduino/issues/10800#issuecomment-720661850).

I'm sure you will find many other opportunities to contribute to the Arduino open source community.