arduino/Arduino #10797: Add a library controlKeyboard

YashuYathi, 2020-09-25:
Please add this Arduino library.
[**https://github.com/YashuYathi/controlKeyboard.git**](url)

per1234, 2020-09-25:
Hi @YashuYathi. In order to be added to the Library Manager index your library must contain a library.properties file. The format of this file is documented here:
https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

After adding the library.properties file, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

YashuYathi, 2020-09-26:
I have done the changes. Please review it.

per1234, 2020-09-26:
The library looks good to me now. However, it was not compliant with the requirements at the commit the 1.0.0 tag was made on:
https://github.com/YashuYathi/controlKeyboard/tree/1.0.0

So you'll need to either create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing `1.0.0` tag.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

YashuYathi, 2020-09-26:
I didn't get it. It is 1.0.0

YashuYathi, 2020-09-26:
Sorry about that. I have changed it.

per1234, 2020-09-26:
Your repository is still not compliant.

Please carefully follow these instructions:
1. Open this link: https://github.com/YashuYathi/controlKeyboard/releases
1. Click the "Draft a new release" button near the top right corner of the page.
1. In the "Tag version" box, type "1.0.1"
1. In the "Target" menu, make sure "master" is selected.
1. Click the "Publish release" button.

YashuYathi, 2020-09-26:
>    2. Click the "Draft a new release" button near the top right corner of
>    3. In the "Tag version" box, type "1.0.1"
>    4. In the "Target" menu, make sure "master" is selected.
>    5. Click the "Publish release" button.

Is this correct?

per1234, 2020-09-26:
Looks good to me. Thanks!

cmaglie, 2020-10-06:
Done, sorry for the long delay.

closed 2020-10-06 by cmaglie