arduino/Arduino #10798: Arduino rewrites my Marlin

mytja, 2020-09-25:
Hello!
I recently compiled a new Marlin for my 3d printer, and when I compiled Arduino rewrote my Configuration.h file, which I worked so hard on it.
I'm using Arduino IDE 1.8.13 on a Windows 7 machine
[Configuration.zip](https://github.com/arduino/Arduino/files/5284649/Configuration.zip)
This is what's left from it.
Thank you

matthijskooijman, 2020-09-25:
Hm, that's weird. I looked at the `Configuration.h` file inside the zip, it's 68kb of zeroes.

I can imagine it's frustrating when this happens after hard work, the IDE really should not cause dataloss like that.

However, I can't really think of any way that a compilation run would cause something like this. No files in your sketch directory are written during the compilation process, everything happens in a temporary directory. I also haven't heard any similar reports so far, @per1234, you maybe?

Can you remember a bit more from when this happened? Have you been editing the file (or other files) in the Arduino IDE before compilation? Did you save before compiling? When did you notice something was broken? During or after compilation in the IDE? Or only after you closed the IDE and started again later maybe? Can you find a way to reproduce the problem on your system?

I'm afraid this is going to be a hard issue to find the cause of. There might be a rare IDE bug involved, but there might also be something specific to your system (maybe a virus scanner doing something weird, or bad hardware maybe?)...

matthijskooijman, 2020-09-25:
Also: Did you make any manual changes to configuration (e.g. Arduino's `preferences.txt`, `theme.txt` or  `platform.txt`)?

And what board were you compiling for?

mytja, 2020-09-25:
Yeah, it's weird.
I compiled new changes with Arduino and then I got this error:
`Can't write to a file` or something similar from Arduino IDE. So I restarted my computer and got presented with a blank script.
I didn't save files after one line modification, just clicked compile.
I wasn't editing files in other programs and I sadly can't reproduce this issue.
Yes, for compiling. I compiled for AVR Marlin board, which was custom, but never had any issues. Got Marlin and AVR board from here: https://www.thingiverse.com/thing:2995932/files

matthijskooijman, 2020-09-25:
Any chance that your disk was full (i.e. is it nearly full now)? That can sometimes also produce weird stuff. I know that Marlin is rather big, so maybe it used a lot (10s mabye 100s of mb) of temp space that caused the disk to be full, and maybe at that time you saved your sketch in the IDE or so?

Or maybe there was some genuine disk I/O problem somehow. Maybe double-check the SMART health data of your disk? :-)

mytja, 2020-09-26:
My C disk is almost full, but, I've got arduino as a zip on D disk and also marlin is on d disk

matthijskooijman, 2020-09-26:
> My C disk is almost full, but, I've got arduino as a zip on D disk and also marlin is on d disk.

Hm. The temporary build dir might very well end up on C and fill up the disk, but even then I would not expect writes to the sketch on D to be messed up.

I think I'm out of ideas, unless you can somehow reproduce the problem (reliably)...

mytja, 2020-09-28:
C disk has approx. 1gb of empty storage and SMART health data says, that my disk is OKAY...

matthijskooijman, 2020-09-28:
Then I'm out of suggestions, I'm afraid.

per1234, 2020-10-18:
There's a similar report here: https://github.com/arduino/Arduino/issues/3121

mytja, 2020-10-19:
Interesting...

mytja, 2020-12-09:
I reproduced this issue!
When I leave my PC for a long time on, i always get some kind of a error: ```Not enough memory to make this operation```, which I mentioned earlier, even though it seems that i have 11GB left. Then, because of this Windows error, Windows thinks I don't have enough memory and seems to overwrite the sketch.
Tell me if you need anything!

matthijskooijman, 2020-12-09:
Interesting. Is that literally the exact error you get, or something similar to that? How does this error appear? Inside the IDE window? Or as a separate popup in the IDE or in Windows? Or something else?

mytja, 2020-12-10:
When I leave my PC running (even without any programs) for a while, I get ```There is not enough memory to complete this operation``` if I translate it from Slovenian language. I can open some apps (like Arduino IDE or Chrome), but Chrome warns me about failure on its start, while Arduino IDE doesn't warn me! Then when I save it, I get a full file of spaces!

It is like that: when I'm doing something with Cura or Android Studio, I get that error, in a popup, but when i have a CMD i get that error directly in CMD! This is definetly Windows related error, then! Will try to uninstall some sketchy apps, and will notify you, if I get succesfull.

Thank you so much!

mytja, 2021-05-04:
This is my computer's fault, since it had corrupted registry.
Closing this issue

closed 2021-05-04 by mytja