arduino/Arduino #10799: AS6212 Temperature Sensor Library Request

will2055, 2020-09-25:
https://github.com/will2055/AS6212-Temperature-Sensor-Breakout/files/5285638/AS6212-Temperature-Breakout.zip

Submission of my temperature sensor library.

will2055, 2020-10-07:
Please re-open:

Arduino library manager isn't installing library correctly. Revised to clean new repo. Please update information in Arduino Library Manager.

**Main repository: https://github.com/will2055/AS6212-Arduino-Library**

Release (v1.0.0): https://github.com/will2055/AS6212-Arduino-Library/releases/tag/v1.0.0

Zip Dowload: https://github.com/will2055/AS6212-Arduino-Library/archive/v1.0.0.zip

cmaglie, 2020-10-17:
I've updated the index, now the new repository is used.

I see that you forget to update the version in the `library.properties`, this is required to publish a new release:
https://github.com/will2055/AS6212-Arduino-Library/blob/v1.0.2/library.properties#L2

To fix this:
1. edit the version in library.properties and set it to 1.0.2
2. commit & push
3. delete and re-do the tag 1.0.2 on the new master

once done, the release 1.0.2 will be automatically published within an hour.

closed 2020-10-17 by cmaglie