arduino/Arduino #10802: jsonlib updated release

wyojustin, 2020-09-26:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
https://github.com/wyolum/jsonlib/releases/tag/0.1.2

per1234, 2020-09-26:
Hi @wyojustin. Your library version at the 0.1.2 Git tag has already been added to the Arduino Library Manager index. It's a bit confusing because you had the `version` value in library.properties set to `0.1.1` when you made the `0.1.2` Git tag:
https://github.com/wyolum/jsonlib/blob/0.1.2/library.properties#L2
```
version=0.1.1
```
Library Manager uses this `version` field for versioning. The name of the tag is not used for anything. So the 0.1.2 tag is available from Library Manager as version 0.1.1.

In the future, make sure to always update the library.properties `version` value before creating a Git tag in your repository.

The Library Manager indexer checks all the libraries on the list for new tags/releases hourly. After that, it may take some time for the updated index to propagate through the CDN cache to make your new release available everywhere, but it will get there in good time.

So you don't need to create an issue report every time you release a new version; once you push a tag/release that is compliant with all the Library Manager requirements everything happens automagically from there!

Reference:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

closed 2020-09-26 by per1234

wyojustin, 2020-09-26:
Thank you.  I forgot to push the change.

Justin