arduino/Arduino #10804: avr-as fails to link to (existing) shared objects

hyphenrf, 2020-09-26:
I don't remember exactly what changed; I had a working arduino IDE before today, but right now even compiling the bundled examples does not work. Instead I get this error:

```
usr/share/avr/bin/avr-as: error while loading shared libraries: libopcodes-2.32.so: cannot open shared object file: No such file or directory
```
<details>
<summary> Full verbose output </summary>
<pre>
Arduino: 1.8.13 (Linux), Board: "Arduino Nano, ATmega328P"

/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -tools /usr/share/arduino/tools-builder -libraries /home/hyphen/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10813 -build-path /tmp/arduino_build_599692 -warnings=all -build-cache /tmp/arduino_cache_69330 -prefs=build.warn_data_percentage=75 -verbose /usr/share/arduino/examples/01.Basics/Blink/Blink.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -tools /usr/share/arduino/tools-builder -libraries /home/hyphen/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10813 -build-path /tmp/arduino_build_599692 -warnings=all -build-cache /tmp/arduino_cache_69330 -prefs=build.warn_data_percentage=75 -verbose /usr/share/arduino/examples/01.Basics/Blink/Blink.ino
Using board 'nano' from platform in folder: /usr/share/arduino/hardware/arduino/avr
Using core 'arduino' from platform in folder: /usr/share/arduino/hardware/arduino/avr
Detecting libraries used...
/usr/share/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/eightanaloginputs /tmp/arduino_build_599692/sketch/Blink.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/usr/share/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/eightanaloginputs /tmp/arduino_build_599692/sketch/Blink.ino.cpp -o /tmp/arduino_build_599692/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
/usr/share/arduino/tools-builder/ctags/ctags-5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_599692/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/usr/share/avr/bin/avr-g++ -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/eightanaloginputs /tmp/arduino_build_599692/sketch/Blink.ino.cpp -o /tmp/arduino_build_599692/sketch/Blink.ino.cpp.o
/usr/share/avr/bin/avr-as: error while loading shared libraries: libopcodes-2.32.so: cannot open shared object file: No such file or directory
exit status 1
Error compiling for board Arduino Nano.
</pre>
</details>

Additionally, I did check the availability of library shared objects and LD_LIBRARY_PATH so I'm not sure where else to look:
```
% locate libopcodes
/usr/lib64/libopcodes-2.34.so
/usr/lib64/libopcodes.so
/usr/share/avr/lib/libopcodes-2.32.so
/usr/share/avr/lib/libopcodes.a
/usr/share/avr/lib/libopcodes.so
% echo $LD_LIBRARY_PATH 
:/usr/share/avr/lib
% echo $AVR_TOOLCHAIN_ROOT 
/usr/share/avr
```

per1234, 2020-09-27:
Did you install the Arduino IDE from a package manager, or did you get the official Arduino IDE from https://www.arduino.cc/en/Main/Software?

hyphenrf, 2020-09-28:
My distro's package manager; source of the package is here:
https://dev.getsol.us/source/arduino

hyphenrf, 2020-09-28:
I would like to mention though that the package hasn't changed between me having a working Arduino IDE and a not-working one.
Last update to the package was on *Jun 22*

What did change is me doing a $HOME cleanup that deleted .arduino15 directory, if that may be relevant (a setting that pointed correctly at the location of bundled avr shared objects maybe?)

per1234, 2020-09-28:
Please try the official Arduino IDE and then let us know whether the problem still occurs.

We don't provide support for the unofficial package manager versions.

hyphenrf, 2020-09-28:
I understand where your response is coming from but please consider:
1. none of the relevant packages (`arduino` , `avr*`) changed at all between having a working and non-working IDE. It's highly unlikely that it's some misbehaving upgrade.  
2. the package is just re-packaging of your official sources, built and conveniently handled for upgrades etc. You can easily verify that because the build recipe is publicly available under the link I sent. but that may be irrelevant to begin with as said in point 1.  
3. it may be more useful for me (and anyone in the future seeing this issue) to point me to a resource on how the underlying build tools discover their shared libraries and whether the IDE, or `~/.arduino15` have a role to play in this.. Instead of the catch-all solution of just installing the official binary dist. I don't mind putting more effort myself into pinpointing the actual problem and helping others who may end up in a similar situation.  

lmk what you think

facchinm, 2020-09-28:
@hyphenrf you should notify your distro maintainers about the missing recompilation of `avr-gcc` package after updating  `libopcodes` (or you partially updated the system, so a full update will solve it on your side).

Said that, we can't give any support for unofficial repackaging of any of our tools. Official binaries are compiled in a way they need minimal baseline dependencies to run on a very large set of Linux distributions.

closed 2020-09-28 by facchinm

matthijskooijman, 2020-09-28:
@hyphenrf, I can see your points, but it is simply not doable for the Arduino dev team to *also* dive into each distro-modified version of Arduino to provide support for them. Hence the question: If this also occurs with the official Arduino version, then this is the place to get support, if not, your distro should be the point of contact.

That does not necessarily mean that there is no problem in the IDE code itself, but if there is, further discussion a  the distro level might point out the actual cause and if it *is* a problem in the IDE (which would then be exposed by the way the distro packages the IDE), a more specific bug report can be opened and things can be fixed.

If you're interested in debugging this, some things that might be helpful:
 - Normally, the IDE includes a copy of e.g. `avr-gcc` and related tools. In your distro installation, changes have been made to the bundled AVR core, or the IDE, to let the IDE use the system's `avr-gcc` rather than a bundled one. This makes sense from a distro perspective, but is hard to support from Arduino's perspective (also, cores installed through the board manager will typically bring their own compilers anyway, depending on how the distro implemented this).
 - Given that system-wide `avr-as` is used, there's not much that the IDE can really do about a library error like shown. It's likely that the `avr-gcc` / `avr-as` package is simply broken in your distro (as suggested by @facchinm). You can probably test this easily by just running `avr-as` outside of the IDE, if that also breaks, this has nothing to do with the IDE or the IDE distro packaging at all.
 - It's weird that this didn't happen with your old .arduino15 directory, but if you do not have a copy of that working .arduino15 dir to compare, this is not really helpful. The only reason I can think of that this would influence things, is that before you had a different version of the AVR core installed through the board manager. This version ends up in .arduino15, normally (if your distro has not patched the IDE in this area) with a copy of `avr-gcc` if it is default from the bundled copy. So maybe you were using `avr-as` from the AVR core before, and removing .arduino15 reverted back to the  system version, exposing a bug in there.

chax, 2020-09-30:
I am official maintainer of this package in Solus repo and I can confirm that this is packaging issue which will be resolved.

matthijskooijman, 2020-09-30:
@chax, cool, thanks for following up!