arduino/Arduino #10818: RAM disk

PaulVdBo, 2020-09-30:
I wonder if it possible to run the whole IDE from a RAMdisk, or at least use a RAMdisk for compiling. I see while compiling that there's a lot of reads and writes to HD
Of course, it wouldn't make much difference if the program is small. But I guess it could save a lot of compiling time when the sketch is a few thousand lines.

[image]

Thanks in advance. :)

matthijskooijman, 2020-09-30:
I would expect your OS to handle this with caching already. On linux, any files that have been recently read or written, will have their contents present in the page cache, so all reads and writes can happen from RAM already (unless an application explicitly calls something like fsync to force a write to disk, but I wouldn't expect gcc to do so). I assume Windows and OSX do something similar.

You could force this to be in RAM, but that's fairly non-trivial for the IDE to do (requiring OS-specific setup, possibly requiring root/admin privileges) and this would take away RAM from the OS cache, meaning it could even result in *worse* performance instead of better (at in some cases, e.g. when memory is scarce).

If you want to try this, you can create a ramdisk and set `build.path` (in `preferences.txt`) to a path inside the ramdisk, and see if that improves performance for you. If you want to try this, be sure to report the results. If not, maybe you should close this issue.

PaulVdBo, 2020-09-30:
Hey Matthijs,
Thanks for your fast reply. I will try your the "build.path" solution and will come back with the results soon.

PaulVdBo, 2020-09-30:
Here iz ze rezultz ov ze jury : 
I restarted the IDE after every change and measured time from start compiling till start uploading.
a program with about 2500 lines compiled about 5 seconds faster when I use : "build.path=R:/ArduinoTemp" (R:/ being my Ramdisk) 30seconds instead of 35 seconds....
So worth it ? I would say "yes" if you have to compile and recompile and recompile and re... a "BIG" sketch ...
_Please don't tell my boss !!! He'll think I can do MORE stuff in an hour !_ ;)

Anyhow : THANKS for the valuable tip !

matthijskooijman, 2020-09-30:
Did you compare with and without build.path, or both cases with build.path on and off ramdisk? Since if you omit build.path, then it will start with an empty build dir, while if you use build.path, it will be kept between starts of the IDE, so you might be using some cached files from a previous run (though at 30s, it sounds like the RAMdisk approach might still be useful).

Regardless, even though 35->30s is signficant, I do not think it's feasible to implement this in the IDE, so I'm going to close this ticket (but feel free to add more comments).

PaulStoffregen, 2020-09-30:
If you're looking for speedup ideas, something I tried on Teensy which made a small improvement is a precompiled Arduino.h.gch header.  Here's the platform.txt recipe we're currently using on Teensy.

```
## Precompile Arduino.h header
recipe.hooks.sketch.prebuild.1.pattern="{compiler.path}precompile_helper" "{runtime.platform.path}/cores/{build.core}" "{build.path}" "{compiler.path}{build.toolchain}{build.command.g++}" -x c++-header {build.flags.optimize} {build.flags.common} {build.flags.dep} {build.flags.cpp} {build.flags.cpu} {build.flags.defs} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DF_CPU={build.fcpu} -D{build.usbtype} -DLAYOUT_{build.keylayout} "-I{runtime.platform.path}/cores/{build.core}" "{build.path}/pch/Arduino.h" -o "{build.path}/pch/Arduino.h.gch"
```

This only helps when a file includes Arduino.h first.  Many libraries do, but not all.  Of course I've updated many of the ones we use on Teensy to include Arduino.h first, because we're using this in the build.

Maybe this is too off-topic.  If so, please forgive me.

closed 2020-10-02 by per1234