arduino/Arduino #10819: New search not comprehensive enough

BallscrewBob, 2020-09-30:
Would it be possible for the search to also cover https://forum.arduino.cc/index.php?board=118.0 ?
I did not see it in the list from any of the drop downs.

matthijskooijman, 2020-09-30:
What search are you referring to exactly? The search on the forum? Can you provide an example of a search that does not return what you expect it to?

If you do mean the forum, I'm pretty sure this is not the right place about this (but I'm not so sure what the right place is, though).

BallscrewBob, 2020-09-30:
Having checked I am sure this is the right place as it is a search issue and not a specific forum issue.
It also seems to have been repaired as search results now include the section I linked to
The actual section I linked to is however not **listed** so it does need to be added.

[image]

matthijskooijman, 2020-09-30:
> Having checked I am sure this is the right place as it is a search issue and not a specific forum issue.

Well, this repository is primarily about the Arduino IDE software, though it also serves as a bit of a catchall for other stuff like the website sometimes. So maybe this is the right place.

Ok, with the info you gave, I managed to puzzle together what I think you mean.

When you do a search on the website or forum, then select Forum, you can select forum categories and board on the left. If you select the "Using Arduino" category, and then the "Forum Boards" dropdown shows boards:

https://www.arduino.cc/search?q=foo&tab=forum&menu%5Bcategories%5D=Using%20Arduino&page=1

However, it does not actually show all boards (it seems "Installation & Troubleshooting", "Introductory Tutorials" and "Avrdude, stk500, Bootloader issues" are missing).

If this is indeed the list you were referring to, then I think this should be enough info for the webmasters to look into this (not sure who or when will do that, though).

BallscrewBob, 2020-09-30:
Yes that is the one. 
You have hit the nail on the head.
I was advised to use this section as it was a WEB issue and not a forum specific one by an Arduino employee / contractor.

Thanks and sorry if I made it hard for you.
Well done spotting the other missing sections too !