arduino/Arduino #10823: [Library Manager] Please add Ticker library

vzahradnik, 2020-10-01:
Please add https://github.com/JSC-electronics/Ticker library.

Thanks,
Vladimir

per1234, 2020-10-01:
Hi @vzahradnik. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value used in your library has already been taken:
https://github.com/sstaub/Ticker

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

vzahradnik, 2020-10-01:
@per1234 I changed the name to `SimpleTicker`, and I also created a new release.

Please review the `library.properties`, and let me know if there's anything else to be done.

Thanks!

https://github.com/JSC-electronics/Ticker/blob/master/library.properties

cmaglie, 2020-10-06:
Done.

closed 2020-10-06 by cmaglie