arduino/Arduino #10824: [Library Manager] Please update the Blues Wireless Notecard library

bsatrom, 2020-10-01:
Library Source: https://github.com/blues/note-arduino

Tagged Release: https://github.com/blues/note-arduino/releases/tag/v1.2.4

Zip Direct download: https://github.com/blues/note-arduino/archive/v1.2.4.zip

per1234, 2020-10-01:
The Library Manager indexer checks all the libraries on the list for new tags/releases hourly. After that, it may take some time for the updated index to propagate through the CDN cache to make your new release available everywhere, but it will get there in good time.

So you don't need to create an issue report every time you release a new version; once you push a tag/release that is compliant with all the Library Manager requirements everything happens automagically from there.

Reference:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

closed 2020-10-01 by per1234

bsatrom, 2020-10-01:
ah, great to know, thanks!