arduino/Arduino #10831: [Library Manager] please remove the TelnetStream2 library

jandrassy, 2020-10-03:
someone published my library before I did. and changed the author.

per1234, 2020-10-03:
If I understand correctly, these are the libraries in question:

* The library already in the index: https://github.com/ameer1234567890/TelnetStream2
* @jandrassy's library: https://github.com/jandrassy/TelnetStream

@jandrassy I think you and @ameer1234567890 are going to need to work this out between yourselves. I see you already have started that discussion: https://github.com/ameer1234567890/TelnetStream2/issues/2, which I am now following, so please confine the further discussion over there. I'm sure you can come to an agreement. Once that is concluded, feel free to come back here to request any removals or inclusions to the index.

Arduino can't be responsible for micromanaging the thousands of libraries in the Library Manager index. If there is a case of malicious or illegal behavior, certainly we will investigate, but I just don't see any evidence of that in this case.

closed 2020-10-03 by per1234

jandrassy, 2020-10-03:
Per, my library is in Library Manager too

ameer1234567890, 2020-10-03:
@jandrassy Instead of hostility and calling each other's version "garbage", let's settle this elsewhere.

jandrassy, 2020-10-03:
> @jandrassy Instead of hostility and calling each other's version "garbage", let's settle this elsewhere.

not the library is garbage. redundant old version is.
you didn't change anything so why would I call my library garbage?