arduino/Arduino #10833: inside IDE Copy Info to Cliboard

StefanL38, 2020-10-04:
newbees tend to ask quick questions with insufficient information about their setup. So a function that collects important data like:

- selected board 
- IDE-Version, 
- selected COM-Port,  
- COM-Port-driver
- operating system
 - include-files of opened *.INO-file

and copies this info to the clipboard 
ready to paste into a posting would improve the process of giving helpful answers.

It would reduce the all-day ping-pong-game of 

`while (NewBeeDoesKnowVeryLittle) {
  newbee posts (insuficient) question
  helpers ask back 
  newbee posts (still insuficient) question
}
`
getting the IDE's Version and the selected board should vrey easy

best regards Stefan

per1234, 2020-10-06:
The IDE's "Copy error messages" button adds the IDE version, OS, and board name to the copied output. However, that button is only available when there is an error. Perhaps it would sometimes be useful to have access to this information when there is no error (code compiles and uploads, but is not working as the user expects).

Related: https://github.com/arduino/Arduino/issues/5398

PaulStoffregen, 2020-10-06:
The original UI intention was for the button to appear where none has previously been present to serve as a gentle suggestion to the user that they should copy the entire error message when seeking help.

Before we added this button (I contributed the original code) the Arduino forums rarely had copies of actual error messages.  People may have realized they could highlight text and use CTRL-C (CMD-C on Mac) to copy the text, but hardly anyone actually did.  The button's purpose isn't just to make copying text easier, but to visually suggest to users that they should copy the error messages.

If the button is made to be always visible, please consider some sort of visual effect to draw the user's attention to the button when their Verify or Upload results in an error.