arduino/Arduino #10835: [Feature request] make EditorConsole.scrollDown() smart

tobozo, 2020-10-06:
Summary : `EditorConsole.scrollDown()` creates a UX problem when the preference(s) `Show verbose output during compilation/upload` is/are enabled.

How to reproduce: 

- Choose a big sketch with many library depedencies that will produce enough compliation statements to trigger a console scroll.
- Add a #warning somewhere in the code
- Attempt to read the warning while it's compiling

Problem observed: It's impossible to lock the console scroll on a specific line while the IDE is compiling

Possible solution: temporarily neutralize auto scrolldown on user interaction with the console, eventually restore on scroll-end (native behaviour of linux console)

This feature request is about how to avoid fighting with the scrollbar whenever this occurs.

Proposed behaviour is to only affect the scrollDown behaviour *during compilation* and only when the `Show verbose output during compilation/upload` option is enabled.

It does not need to have have a persistent effect.

Possible UI triggers for disabling automatic scrolldown : 
  - Focusing on the console
  - Mouse wheel / Touch swipe
  - Scrollbar interaction
  - First warning/error detected

Possible triggers for cancelling  : 
  - Scrolling down to the bottom
  - Pushing the "end" of "pgdown" key

matthijskooijman, 2020-10-06:
A common approach for this is to simply only do autoscrolling when the window is already in the bottommost scroll position (before inserting new text). This is slightly less smart than what you propose, but would work well in most cases (might need you to scroll once more than with your suggestion) and is probably a lot easier to implement and more predictable.

Also note #6081, which implements a related (but different and independent) feature where autoscrolling stops at the first error or warning shown (but that turned out to have some caveats and is not a proper implementation yet).

tobozo, 2020-10-06:
Thanks for your feedback!

#6081 is about minimizing a side effect by providing an empathic (although non standard) behaviour, this is a perfectly valid solution to the symptom.

However it's still out of control since it's based on the assumption that only warnings and errors are of interest, plus it must parse the output to figure it out, which raises risks of false positives and adds processing overhead.

This is why using cognitive inheritance from linux console behaviour just seems a better UX solution to that problem.

Since I'm only suggesting the feature I'm okay with both solutions anyway, just let me know if this one redundant and I'll close the issue.