arduino/Arduino #10836: Support for macOS Big Sur on arm64

neilpa-inv, 2020-10-06:
With the ["Apple Silicon" announcement](https://www.apple.com/newsroom/2020/06/apple-announces-mac-transition-to-apple-silicon/) earlier this year, are there any plans for macOS/arm64 support? As I understand it this at least requires updates to the avr-toolchain and `avrdude` along with other stuff I may not be aware of.

My primary interest is in getting `avrdude` working on an arm64 Mac. I've got access to one of the dev kit machines and have managed to compile a version on it. However, I've yet to successfully flash an Arduino from with the same parameters I've used from an Intel Mac.

I realize that some of the above may be better suited to the [arduino/avrdude-build-script](https://github.com/arduino/avrdude-build-script) repo (which I just discovered), but I'm also generally interested in the Arduino IDE on this upcoming platform.

ubidefeo, 2020-10-21:
hi @neilpa-inv 

we have of course considered this, and got our hands on a dev kit.
In the initial phase of the transition everything should keep working thanks to Rosetta 2, but in the future we should have native versions of every bit of software we release.
Of course we cannot account for 3rd party tools which are simply installed via our Boards Manager, but when it comes to binaries we build internally we'll make sure they are running properly on every major platform.

brandwe, 2020-11-24:
I have tried this and want to report that using the opensdk build by Azul [using these instructions](https://gist.github.com/tresf/d3ea928895320f2f44196f270d8ef9f0#file-openjdk-stopgap-sh-L13) I still get an error after successful compile. I'll post back as I find more, but I'm sure it has to do with some x86 binaries in the keychain.

portable-false:

generate-hourly-build-txt:

start:

macosx-start:
     [exec] 2020-11-23 22:24:44.730 Arduino[24610:896895] Loading Application 'Arduino'
     [exec] 2020-11-23 22:24:44.731 Arduino[24610:896895] JVMRuntime=zulu-8.jdk
     [exec] 2020-11-23 22:24:44.731 Arduino[24610:896895] CFBundleName=Arduino
     [exec] 2020-11-23 22:24:44.731 Arduino[24610:896895] WorkingDirectory=(null)
     [exec] 2020-11-23 22:24:44.731 Arduino[24610:896895] JVMMainClassName=processing.app.Base
     [exec] 2020-11-23 22:24:44.731 Arduino[24610:896895] JVMOptions=(
     [exec]     "-Dapple.awt.application.name=Arduino",
     [exec]     "-Dcom.apple.macos.use-file-dialog-packages=true",
     [exec]     "-Dcom.apple.smallTabs=true",
     [exec]     "-DAPP_DIR=$APP_ROOT/Contents/Java",
     [exec]     "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
     [exec]     "-Djava.net.preferIPv4Stack=true",
     [exec]     "-Xdock:name=Arduino",
     [exec]     "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
     [exec]     "-Dfile.encoding=UTF-8",
     [exec]     "-Xms128M",
     [exec]     "-Xmx512M",
     [exec]     "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
     [exec] )
     [exec] 2020-11-23 22:24:44.731 Arduino[24610:896895] JVMArguments=(
     [exec] )
     [exec] 2020-11-23 22:24:44.732 Arduino[24610:896895] JVMClasspath=(null)
     [exec] 2020-11-23 22:24:44.732 Arduino[24610:896895] JVMDefaultOptions={
     [exec] }
     [exec] 2020-11-23 22:24:44.732 Arduino[24610:896895] SearchSystemJVM=false
     [exec] 2020-11-23 22:24:44.732 Arduino[24610:896895] -> Bundle path: /Volumes/Development/Electronics/Arduino/build/macosx/work/Arduino.app
     [exec] 2020-11-23 22:24:44.732 Arduino[24610:896895] -> Working Directory: '/Volumes/Development/Electronics/Arduino/build'
     [exec] 2020-11-23 22:24:44.732 Arduino[24610:896895] -> JVM Runtime path: /Volumes/Development/Electronics/Arduino/build/macosx/work/Arduino.app/Contents/PlugIns/zulu-8.jdk
     [exec] 2020-11-23 22:24:44.732 Arduino[24610:896895] -> Java Runtime Dylib Path: '/Volumes/Development/Electronics/Arduino/build/macosx/work/Arduino.app/Contents/PlugIns/zulu-8.jdk/Contents/Home/jre/lib/jli/libjli.dylib'

BUILD FAILED
/Volumes/Development/Electronics/Arduino/build/build.xml:146: The following error occurred while executing this line:
/Volumes/Development/Electronics/Arduino/build/build.xml:543: exec returned: 1

adeeb10abbas, 2021-01-13:
Just got a new M1 mac, and would love to help you guys out if possible! Arduino was what got me into coding! I am a 3rd-year Computer Engineering student. What would be a good point to start working on this issue! Thanks!

Grippy98, 2021-02-04:
@brandwe Well I got it to build just fine on first try, but alas "Error loading Java on startup"

[image]

parreirat, 2021-02-18:
I've been scouring the internet for the past couple hours trying to get M1 to detect my Mega2560 R3. Does not show up in serial ports at all. Have rebooted repeatedly, replugged, tried different cables, usb hubs.
OSX is not even recognizing an "Unknown device" or similar: cannot detect any trace of board from PC.
Tried from Arduino IDE Pro, Arduino IDE, arduino-cli with `arduino-cli board list`

Tried installing drivers listed here, but did not help (and figures: it's listed for 10.9 up, so my hopes were low)
https://learn.sparkfun.com/tutorials/how-to-install-ftdi-drivers/all#mac

Grippy98, 2021-02-18:
Very curious, just tried with both a genuine and a clone Mega 2560 R3 just now and both uploaded fine. What version of Arduino are you running? 

[image]

parreirat, 2021-02-18:
x<<

> Very curious, just tried with both a genuine and a clone Mega 2560 R3 just now and both uploaded fine. What version of Arduino are you running?
> 
> [image]
> 
> I should add it shows up as a "Communications Device" as you can see on the bottom right.

Godammit, can't believe it works for you.

I'm also on 11.2.1, and arduino 1.8.13 as well... I just don't get it showing up as a port. 
USV shows nothing either.

How are you running the board? I have the USB-B cable that came with it hooked onto my external USB C/A hub.

Currently trying to set up GO + build this PR from arduino-user-agent that claims to add M1 support (experimentar, not really a lot of info, but I found it):
https://github.com/arduino/arduino-create-agent/pull/617

Grippy98, 2021-02-18:
I assume that "scarlet" is a USB Audio interface? Try that port?

parreirat, 2021-02-19:
> I assume that "scarlet" is a USB Audio interface? Try that port?

I had already tried the board on all 3 ports of my hub with everything else disconnected - did not help

neilpa-inv, 2021-02-19:
> I had already tried the board on all 3 ports of my hub with everything else disconnected

@parreirat this sounds like a hardware problem with the Arduino or the cable/hub. Especially since it's not showing up in the USB device tree.

> I just get this "composite device" which I don't think is it, since it's there even if I unplug the board.

Correct. You can see in the screenshot that @Grippy98 shared it's listed as "Communications Device" with different vendor and product IDs. You can also see Arduino under the manufacturer field.

parreirat, 2021-02-19:
> @parreirat this sounds like a hardware problem with the Arduino or the cable/hub. Especially since it's not showing up in the USB device tree.
>
>
> Correct. You can see in the screenshot that @Grippy98 [...]

So you think it's more of a problem with hub/cable or arduino...
I have a spare USB B from my e-drums - and I'll ask a friend for his USB HUB, that way I can eliminate if the arduino itself is the issue.

I'll report back. Thanks!

parreirat, 2021-02-19:
It was the USB B cable. I used another one and it's working flawlessly now :)

dinolupo, 2021-05-26:
@Grippy98 yes thank you, same error on M1 unfortunately.

PaulStoffregen, 2021-05-28:
It's not just M1.  Compiling for Intel Edison boards doesn't work on Intel-based Macintosh either, if running any version of MacOS newer than Mojave.

[image]

As you can see in the screenshot, the problem is the toolchain in the "Arduino i686 Boards" package is a 32 bit executable.  Apple ended support for 32 bit programs starting with Catalina.

dinolupo, 2021-06-02:
@PaulStoffregen ok thank you. I can imagine that there is no solution at this point, besides using another O.S.

PaulStoffregen, 2021-06-03:
In theory, anyone could try to update the "Arduino i686 Boards" package.  As far as I know, this is _only_ the gcc toolchain which should be open source.  With enough effort, there should be a way to rebuild the toolchain as x86-64 code.

In practice, it's hard to imagine anyone would invest that sort of effort to support a board Intel discontinued years ago.  And if anyone outside Arduino & Intel were to do it, whether Arduino would publish the updated the package might also be a good question.

imrehorvath, 2022-08-16:
Rosetta 2 can be used as a workaround for now, but a native arm64 built tools like `avr-gcc`, `avrdude` and `arduinoOTA` would be niece to have on Apple silicon Macs. I know the `arduino-cli` has already been prepared to use the best match from the available packages and now falls back to the x86_64 on Apple silicon Macs. What remains is to have these tools compiled for the arm64 architecture.