arduino/Arduino #10839: Arduino.cc is down emailadress to inform webmaster

StefanL38, 2020-10-07:
right now 07.10.2020 09:02 AM the website  Arduino.cc is not reachable. 
So I can't even lookup an email-adress. What's the best suited Email-adress to inform the webmasters of the Arduino-website about problems with the site?

best regards Stefan

endorama, 2020-10-07:
Hello @StefanL38, I'm Edoardo a DevOps Engineer at Arduino. This repository is monitored (thanks @per1234 for the quick triage) so it's a proper place. You may also use the `webmaster@arduino.cc` email address in the future.

Getting back to the website, where you trying to reach `arduino.cc` or `www.arduino.cc`? How long the downtime lasted, can you give us an estimation if not a precise timing? Our monitoring does not show any sign of misbehaviour, so I'd like to dig deeper in this issue.

The site should be regularly online now.

Thanks!

StefanL38, 2020-10-07:
Hi thanks for the email-adress. Now that Ive read it pretty obvious ;-))

I was outside until 12:30. 
good that you aksed back which side exactly. usually I directly load
https://forum.arduino.cc/index.php?board=4.0  (User-Forum Subforum programming questions)

now the site is loading
best regards Stefan

closed 2020-10-07 by StefanL38