arduino/Arduino #10843: Serial Monitor output lags

ropg, 2020-10-08:
I connect with Arduino 1.8.13 on MacOS Mojave 10.14.6 to an M5Stack Core2 ESP32 device. With the Serial Monitor open, if there is no data out of the M5Stack for a little while and then there is, it takes a while for the Serial Monitor to realise there is data. When it does wake up it displays the data all in one go. Sometimes it's just a second or two, sometimes it's longer (10 seconds?). It always updates immediately if I move the mouse pointer.

ubidefeo, 2020-10-08:
hi @ropg 
I dont' have a Core 2 but I bet the original M5 Stack will work the same.
Care to share your code so I can test it on my machine?
I also run Mojave 10.14.6 on my laptop

ropg, 2020-10-08:
It is the same on the other M5Stack which I used before. In fact it's the same on any serial output, I would think. It does not seem dependent on what code I use and think this could be shown with anything that produces serial output but has long enough pauses in it to let the serial monitor "fall asleep", so to speak.

ubidefeo, 2020-10-08:
@ropg 
I have to admit that I tend to use other means of connecting to serial ports, either `screen` or Serial 2, but I'll give this a go and let you know :)

ropg, 2020-10-08:
Just tested: even this, on an M5Stack classic, does the trick:

```C++
#include <M5Core2.h>

void setup() {
	M5.begin();
}

void loop() {
  M5.update();
  if (M5.BtnA.wasPressed()) Serial.printf("millis(): %d\n", millis());
}
```

ubidefeo, 2020-10-09:
@ropg 
with "does the trick" you mean "exhibits the issue", right?
I'm in my studio later today so I can check it out on the M5

ropg, 2020-10-09:
Yes, if you let your computer sit for a while (differs how long that needs to be for me) and then press button it can take up to 10 seconds for data to appear in monitor.

ubidefeo, 2020-10-11:
@ropg 
I grabbed my M5 Stack and ran a few tests, first with my usual tools (Serial app for Mac) and then I am now gonna test in the IDE.
As I started to use it I thought about default UART speeds in the M5 lib when you run `M5.begin()`.
The speed is automatically set to 115200, while the Serial Monitor defaults to 9600.
Have you set the speed to that?
I'll keep you posted on the results anyway ✌️ 

u.

ubidefeo, 2020-10-11:
@ropg 
- left the laptop go to sleep while doing the dishes
- came back, woke it up
- TouchID logged into it
- tabbed to Arduino IDE's Serial Monitor
- pressed button A 

there it was, immediately my Serial print data
I cannot reproduce your context, I'm afraid.
Maybe the first M5 Stack's library does something different than the Core 2, I don't know.
I did order a Core 2 just for the kicks, though :D

ubidefeo, 2020-10-11:
2 hours later it still receives data as soon as I press the button