arduino/Arduino #10858: New versions of library not updating in library manager

KravitzLab, 2020-10-15:
My library "FED3" is is not updating in the library manager when I issue a new release.  Here is the URL to the library:
https://github.com/KravitzLabDevices/FED3_library

I think the issue may have been that I had the URL incorrect in my previous releases.  Can this field be updated to the above URL to see if this causes the library manager to pick up my new releases?  Thanks!

per1234, 2020-10-15:
Hi @KravitzLab. The reason is that you have a `.development` flag file in your repository. Any library release which has this file in its root folder will be ignored by the Arduino Library Manager. So you only need to remove this file and then either make a release/tag or else redo the existing release/tag.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

closed 2020-10-15 by per1234

KravitzLab, 2020-10-15:
Oh thanks for the info, I'll remove it!