arduino/Arduino #10859: Wrong link for Sketch tutorial

ahmetuludag, 2020-10-15:
I was recommending the tutorials to our intern and realized that there's a wrong link in this page:
https://www.arduino.cc/en/Tutorial/Foundations

Under "Basics" section, there's a "Sketch" link for its tutorial and it should be "/en/Tutorial/Sketch" instead of "en/Tutorial/Sketch"

closed 2020-10-22 by kengdahl

ahmetuludag, 2020-10-22:
@kengdahl it's not fixed though it's closed ?

per1234, 2020-10-22:
Hi @ahmetuludag. The error has just been fixed in the documentation source content. The PR that fixed it (in a private repository, so not visible) was linked to this issue report, so the issue was automatically closed when that PR was merged. The fix won't appear in the documentation on arduino.cc until the fixed documentation source has been published, which will take some time, but it is certainly in progress.

Thanks so much for reporting this error!