arduino/Arduino #10861: [Library Manager] Please add libraries

hideakitai, 2020-10-15:
Hello, I created some libraries. 
Could you add following to a list of libraries ?

HyperDeck
https://github.com/hideakitai/HyperDeck

Thanks in advance.

cmaglie, 2020-10-17:
I've added the library to the indexer but there is a problem becuase symlinks are not allowed:

```
Error while zipping library: Symlink not allowed: github.com/lbernstone/plotutils/src/graph/fontlist.h -> ../common/fontlist.h
```

to fix this please remove the symlink (replacing it with a copy of the file or by #including `common/fontlist.h` from `graph/fontlist.h`) and do another release of the library. Do not forget to update the version in `library.properties` and to commit&push it before making a new git tag, I'm telling you because this is often forgotten.

Once done the new release will be automatically published within an hour. I'm going ahead and closing this issue, please comment here if you need further assistance.

cmaglie, 2020-10-17:
Sorry, disregard the previous comment, I commented on the wrong issue.

The library has been successfully added to the index!

closed 2020-10-17 by cmaglie