arduino/Arduino #10868: Please add the ft817 library

stdevPavelmc, 2020-10-19:
Hi, please add the library ft817 to the Arduino library manager.

This library is for CAT control of the Yaesu FT-817 amateur radio transceiver, this lib has extra features over the normal CAT control defined by the OEM of the radio as it does direct EEPROM access via CAT commands, in fact extending the capabilities way beyond the default set.

The URL of the repository is this: https://github.com/stdevPavelmc/ft817 

This library is needed to better support the Andy's [FT817 Buddy](https://github.com/g7uhn/ft817_buddy) FT817 companion gadget.

Thanks. (First release is a matter of minutes out)

stdevPavelmc, 2020-10-19:
First working release on out!

cmaglie, 2020-10-29:
Done.

closed 2020-10-29 by cmaglie