arduino/Arduino #10876: Created a batch scripts that create a new "boards.txt" with the boards in alphabetic order.

mats-nk, 2020-10-21:
Hi,
I got annoyed when trying to find the correct board in the "Board manager", as they are not in alphabetic order so...

I created a simple batch file and created snips for each board in a single file, from that I created a new boards.txt.

I just did a fast and stupid approach but I could make it smarter if that would help.

And for programmers.txt I just did it manually as its not that extensive.

PS. 00build.txt should be renamed to .bat (windows batch program, I can do bash as well if needed)

[programmers.txt](https://github.com/arduino/Arduino/files/5414661/programmers.txt)

[BuildBoards.zip](https://github.com/arduino/Arduino/files/5414649/BuildBoards.zip)

[00build.txt](https://github.com/arduino/Arduino/files/5414660/00build.txt)

per1234, 2020-10-22:
Nice work @mats-nk.

You might consider publishing your scripts in a repository in order to share them with any Arduino community members who might find them useful.

closed 2020-10-22 by per1234

mats-nk, 2020-10-22:
per1234, I think you have misunderstood the issue, it's NOT about the script, it's about that the board manager is a mess.