arduino/Arduino #10878: Suggestion: bootloader to be burned before the sketch.

rtek1000, 2020-10-21:
Hello,

I would like to suggest that the uC Attiny85 bootloader be automatically burned before the sketch, when the upload is performed using the programmer.

This uC can operate with 1MHz, 4MHz, 8MHz, 16MHz (PLL) but if the user selects the clock, he also has to burn the bootloader, this makes the process more difficult for a beginner because he has to perform an additional procedure that the great part of the tutorials available on the internet does not explain.

Even if 16MHz is selected "No bootloader", it is necessary to burn the bootloader again if the uC is programmed for a different clock.

Thank you.

[image]

[image]

rtek1000, 2020-10-21:
> If this is the first time you are using a brand new chip, or if you need to change the clock source or BOD settings, you must do "Burn Bootloader" to set the fuses. If you previously used optiboot, and wish to program with ISP instead, or if you want to install optiboot on the chip, you must do "burn bootloader" first as well.

Source: [https://github.com/SpenceKonde/ATTinyCore/blob/master/Programming.md](https://github.com/SpenceKonde/ATTinyCore/blob/master/Programming.md)

per1234, 2020-10-21:
This issue tracker is only for bug reports and feature requests related to official Arduino software. ATTinyCore is a 3rd party project, and so feature requests for ATTinyCore are off topic here.

closed 2020-10-21 by per1234

rtek1000, 2020-10-21:
Ok, but does the nucleus have the option of forcing the IDE to operate in this way? Burn the bootloader before sketching automatically, when uploading using the programmer?

per1234, 2020-10-21:
Yes. The Arduino boards platform system allows any arbitrary command to be run when you do an "Upload using programmer". avrdude allows running multiple operations in the same command, making it even easier. It should only be necessary to merge [`tools.avrdude.erase.pattern`](https://github.com/SpenceKonde/ATTinyCore/blob/1.4.1/platform.txt#L127),  [`tools.avrdude.bootloader.pattern`](https://github.com/SpenceKonde/ATTinyCore/blob/1.4.1/platform.txt#L132), and [`tools.avrdude.program.pattern`](https://github.com/SpenceKonde/ATTinyCore/blob/1.4.1/platform.txt#L123) into a single command.

So I think the Arduino boards platform system already provides everything needed to enable this use case. If you do find that a specific feature of the platform system is needed that hasn't been requested yet, you're welcome to submit a feature request.