arduino/Arduino #10879: Error when replacing a .ino file with a new one in a directory (How very Borges of you)

boltalex389-2nd-account, 2020-10-21:
On Windows (10 Pro), I got the "How very Borges of you" (the infinite folder error) when I was trying to replace a file in the directory.

per1234, 2020-10-21:
1. Create a sketch
1. Save the sketch
1. **File > Save as**
1. Open the sketch folder
1. Click the "Save" button.

You now get the error:
>"How very Borges of you"
You cannot save the sketch into a folder inside itself. This would go on forever.

1. **File > Save as**
1. Open the sketch folder
1. Select one of the files of the sketch
1. Click the "Save" button.

The save now completes successfully.

This is confusing behavior because if the sketch has multiple files, all the files are saved, not just the file you clicked on. The sketch is the entire folder, not only the .ino file, but the IDE's "Save as" and "Open" behavior reinforces the incorrect impression that the .ino file is the sketch.

ubidefeo, 2020-10-22:
> the IDE's "Save as" and "Open" behavior reinforces the incorrect impression that the .ino file is the sketch.
@per1234 , I'd like to address this starting with the ne IDE, but it requires some thinking and testing which should be carried on asap

dsyleixa, 2020-10-25:
"Sketch" is misleading for save/save as...
Wishful would be:
File save/save as (i.e., the current active (e.g., .ino) file in the IDE)
Project save/save as (i.e., the entire project "Sketch" folder)

per1234, 2020-10-25:
Considering this IDE is specifically designed to have a simple UI to provide a friendly environment to beginners, I can't see any benefit to providing the ability to save a single file. It's very important to carefully consider every single element added to the UI to determine whether it's really needed. This is what sets the Arduino IDE apart from all the other IDEs

The sketch is a unit. When I save, I want to save the entire sketch, just as the Arduino IDE has always done.

dsyleixa, 2020-10-25:
I have to disagree.
If I want to save, I want to save either a single file - or - the entire folder with all subfolders and all additional files (e.g., data subdirectory and/or extra .h files)

NameOfTheDragon, 2020-10-25:
> I have to disagree.
> If I want to save, I want to save either a single file - or - the entire folder with all subfolders and all additional files (e.g., data subdirectory and/or extra .h files)

"There's more than one way to skin a cat". Arduino is a somewhat "opinionated framework". I think you have to accept that it is going to force users to work in a particular way that not everyone is going to like. It is what it is, and a lot of non-expert users seem to appreciate "the Arduino way". The moment you have multiple files in your sketch, you have probably already outgrown the Arduino IDE. Given how it works, the error message seems reasonable given what you were trying to do and in context, I don't see it as a problem. I personally don't use the Arduino IDE for exactly that reason - It seems very limiting to me and I don't like the editor or anything about it really. I like to work my own way and I don't really like opinionated frameworks. So I don't use it. You have other options, such as Visual Studio with the VisualMicro extension, or VS-Code with the Arduino extension or using PlatformIO. But given the modus operandi of the Arduino IDE, that error message seems entirely appropriate.

dsyleixa, 2020-10-25:
it's not forbidden to make things better and more flexible, so please put it on the wishlist.

boltalex389-2nd-account, 2020-10-28:
Ok