arduino/Arduino #10885: SDA 20 and SCL 21 of arduino due  not working

Zainraza024, 2020-10-24:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

Zainraza024, 2020-10-24:
Hi i am using arduino due and want to set time with tiny rtc. But I am unable to set any time on that rtc
I changed ardiuno due to arduino mega the code works fine and rtc was set.
The problem is that I want to use this rtc on ardriuno due read the time, since I think the problem is with sda and SCL.  
Does any one know how to resolve this issue.

per1234, 2020-10-24:
Hi @Zainraza024 This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2020-10-24 by per1234