arduino/Arduino #10892: [Library Manager] Please add Telit GE910 GSM library to manager

akkoyun, 2020-10-27:
My library repo is : https://github.com/akkoyun/GE910

per1234, 2020-10-27:
Hi @akkoyun. Thanks for your interest in contributing to the Library Manager index.

Leading zeros on the components of the library.properties `version` value are not permitted. You should use  semver compliant versioning:
https://semver.org/

Please bring your library.properties file into compliance and then either create a new tag/release or else redo the existing one.

After that, please comment here so we'll know we can proceed.

akkoyun, 2020-10-27:
@per1234 thanks for your feed back 
i update release please confirm.
thanks.

per1234, 2020-10-27:
The issue of the leading zeros on the version components remains. The zeros here: **0**1.**0**7.**0**1 are all prohibited by the semver specification:
https://semver.org/#spec-item-2
>A normal version number MUST take the form X.Y.Z where X, Y, and Z are non-negative integers, and **MUST NOT contain leading zeroes**.

I notice you have also introduced another violation of the requirements for inclusion in the Arduino Library Manager index. Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

As before, please bring your library into compliance, tag, then comment here.

akkoyun, 2020-10-27:
Hi,
I edit the library.properties and i think now ok :)

cmaglie, 2020-10-29:
Done.

closed 2020-10-29 by cmaglie