arduino/Arduino #10896: [Library Manager] Request to add a library

mo-seph, 2020-10-27:
Request to add a library at the following URL: https://git.ecdf.ed.ac.uk/design-informatics/di-development-board/DesignInformaticsPCB

Note - it's a GitLab, but not the central setup. If that's not OK, we can figure out mirroring a copy on one of the usual providers. Thanks!

cmaglie, 2020-10-28:
hi @mo-seph 

> Note - it's a GitLab, but not the central setup. If that's not OK, we can figure out mirroring a copy on one of the usual providers.

Yes, please, create a mirror on the main gitlab (or github or bitbucket).

I see that you also use the gitlab "subgroups": we still do not support URL in the format `username/subgroup/repository`, it must be `username/repository`. So you have two alternatives:

1. create a mirror repository without using subgroups
2. if you really want to use subgroups, you can create a repository without subgroups and later move the repository under a subgroup. We can still use the "old" URL without subgroups since gitlab will forward all the requests for us.

The latter is a workaround we found with another user, it works even if a bit laborious. We have planned to add support for gitlab' subgroups on our indexer but, due to the very low demand, not in the short term.

mhamilt, 2020-11-02:
@cmaglie The mirror of this repo is now on github

https://github.com/Edinburgh-College-of-Art/DesignInformaticsPCB

cmaglie, 2020-11-06:
Done.

closed 2020-11-06 by cmaglie