arduino/Arduino #10908: [Library Manager] Request to add new library

protocentralashwin, 2020-10-31:
Please add this new library to the Arduino manager:

https://github.com/Protocentral/protocentral-ads1292r-arduino

Thank you!

per1234, 2020-10-31:
Hi @protocentralashwin. The library.properties `name` value of your library is not compliant with the requirements for addition to the Library Manager index due to the use of disallowed characters (`&`, `/`).

You can see the allowed characters here:
https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

It's not documented in the Arduino Library Specification, but I believe that the use of a name longer than 63 characters is also problematic, since the folder name is based on the library.properties `name` value and the maximum folder name length allowed by the Arduino IDE is 63 character's. I can't say for sure whether the name length issue will block inclusion in the index, but I think it would be a good idea to try to use a shorter name.

So please fix the `name` value and then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

protocentralashwin, 2020-10-31:
Sorry @per1234, i did not realise that. 

I have edited this and made a new release tag. Thank you for getting back.

cmaglie, 2020-11-06:
Done.

closed 2020-11-06 by cmaglie