arduino/Arduino #10910: ArduinoModbus library spelling error

mackrasp, 2020-11-01:
https://www.arduino.cc/en/ArduinoModbus/ArduinoModbus
Error in keywords.txt
HOLIDING_REGISTERS

this results in a compile error when using the parameter definition HOLD_REGISTERS from the Arduino website

per1234, 2020-11-01:
Transferred to https://github.com/arduino-libraries/ArduinoModbus/pull/36

However, I am very confused by your statement:
>this results in a compile error when using the parameter definition HOLD_REGISTERS from the Arduino website

keywords.txt is only used for keyword highlighting in the Arduino IDE. It is purely cosmetic. A misspelled keyword in keywords.txt  can never cause a compilation error. It only means that your code won't have the pretty colors you are expecting in the IDE's editor.

Now if you were to use the misspelled "`HOLIDING_REGISTERS`" in your code, that would indeed cause a compilation error. But you're implying the opposite: that using the correctly spelled "`HOLD_REGISTERS`" somehow results in a compilation error simply because it doesn't match the keyword spelling in keywords.txt. That doesn't make any sense at all to me. If this is really what is happening, please provide detailed instructions for how we can reproduce this bug and I'll investigate to see whether this should be re-opened as a separate issue from the fix at https://github.com/arduino-libraries/ArduinoModbus/pull/36.

mackrasp, 2020-11-01:
Here is 
Arduino web page for requestFrom()
https://www.arduino.cc/en/ArduinoModbus/ModbusClientrequestFrom

The **type** is required. I wish to read multiple holding registers, so I have specified the type parameter as HOLD_REGISTERS and the compile errors.

Parameters
id (slave) - id of target, defaults to 0x00 if not specified
type - type of read to perform, either COILS, DISCRETE_INPUTS, HOLD_REGISTERS, or INPUT_REGISTERS
address - start address to use for operation
nb - number of values to read     

my code that errors
//read registers
    if(!modbusTCPClient.requestFrom(ID, HOLD_REGISTERS, 0x00,10)) {
      Serial.println(F("read fail"));
      Serial.println(modbusTCPClient.lastError());
      delay(1000);       
    }
compile log
/Users/mack/Documents/Arduino/ArduinoModbus_TCP_Client_MKR_WiFi_v1_0/ArduinoModbus_TCP_Client_MKR_WiFi_v1_0.ino:78:41: note: suggested alternative: 'HOLDING_REGISTERS'
     if(!modbusTCPClient.requestFrom(ID, HOLD_REGISTERS, 0x00,10)) {
                                         ^~~~~~~~~~~~~~
                                         HOLDING_REGISTERS
Using library SPI at version 1.0 in folder: /Users/mack/Library/Arduino15/packages/arduino/hardware/samd/1.8.3/libraries/SPI 
Using library WiFiNINA at version 1.4.0 in folder: /Users/mack/Documents/Arduino/libraries/WiFiNINA 
Using library ArduinoRS485 at version 1.0.0 in folder: /Users/mack/Documents/Arduino/libraries/ArduinoRS485 
Using library ArduinoModbus at version 1.0.1 in folder: /Users/mack/Documents/Arduino/libraries/ArduinoModbus 
exit status 1
'HOLD_REGISTERS' was not declared in this scope

per1234, 2020-11-01:
Thanks so much for the clarification! So there are two separate typos in effect here:

As the error message indicates, the macro name in the library is "`HOLDING_REGISTERS`":
https://github.com/arduino-libraries/ArduinoModbus/blob/5bf098b067b5e733087d6c67eba581426d09f4bc/src/ModbusClient.h#L31

In your initial report you pointed out the misspelling of this keyword in keywords.txt, which will be fixed by https://github.com/arduino-libraries/ArduinoModbus/pull/36

But what I missed before now, because it didn't match my search of arduino.cc for "`HOLIDING_REGISTERS`" is that there is a different misspelling of this macro name in the documentation: "`HOLD_REGISTERS`".

So I'll reopen this issue as a report of the error in the ArduinoModbus library's `requestFrom()` reference page.

mackrasp, 2020-11-01:
Thanks again
I confirmed that the HOLD_REGISTERS was a typo error and that the correct macro name should be HOLDING_REGISTERS which will compile correctly but I was confused by the lack of color in the IDE so I looked in the keywords.txt which showed another spelling typo. Now I am wiser and happier.

closed 2022-07-17 by per1234