arduino/Arduino #10913: Request to add new library

MrYsLab, 2020-11-01:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
Hi. Please this library to the library manager:
[https://github.com/MrYsLab/Telemetrix4Arduino](https://github.com/MrYsLab/Telemetrix4Arduino)

Thanks.

MrYsLab, 2020-11-02:
@per1234 
Hi. I was not sure if there was an issue with the library or not, but rechecking the library.properties file, I realized that the URL was incorrect and I have corrected it. If there are any other issues that need correction, please let me know. 
Thanks.

MrYsLab, 2020-11-04:
I have bumped the release number and removed keywords.txt which was unused. Please let me know if I need to make any changes to have this published to the library manager. Thanks.

cmaglie, 2020-11-06:
Done.

closed 2020-11-06 by cmaglie